DiamondLightSource/fastcs #15: Prevent truncation of EPICS PVs

GDYendell, 2023-10-25:
It is easy when generating Attributes programmatically for the generated PVs to be too long and get truncated. I think the limit is 60 characters.

It would be good to define some logic to truncate the names in a sane way, enuring the `_RBV` is preserved. This could be by truncating each individual word, removing vowels, or many other options. [ADGeniCam](https://github.com/areaDetector/ADGenICam/blob/a1239438e4c9526dacf9a0dc927bfc7d8c5f5b61/scripts/makeDb.py#L53) does the former.

It might also show a warning to suggest to the developer that the name should be shortened in a way specific to the naming structure for the device.

GDYendell, 2023-10-25:
@marcelldls do you know if there is a similar limit in Tango? I imagine if there is is it a much bigger number.

marcelldls, 2023-10-25:
A tango attribute has a full name with the following schema: domain/family/member/attribute and according to https://tango-controls.readthedocs.io/en/latest/development/general-guidelines/naming.html the limits are 85/85/85/??. An alias can be defined to a maximum of 255 characters... I cant find specifically what the maximum attribute length is but I can ask in the Tango controls slack when I get added

GDYendell, 2024-10-14:
Fixed in #56

closed 2024-10-14 by GDYendell