DiamondLightSource/fastcs #38: Add support for mbbIn and mbbOut records

GDYendell, 2024-05-17:
Add `allowed_values` field to attributes
Use `ComboBox` for `AttrRW` with `allowed_values`

jsouter, 2024-06-12:
Had a go at this here https://github.com/DiamondLightSource/FastCS/pull/40
Also works for if we have a string parameter with a list of allowed values, but may need a rework, as tries to use a common interface for both cases where we provide a mapping of {<label_name>: <value>} where the label_name is the same as value for strings. Could be simplified to just a list of labels if we enforce that all mbbo/mbbi types will only use the enum values from 0 increasing without skipping any. (e.g. currently we could have {"label0": 0, "label2": 2}

GDYendell, 2024-10-17:
Fixed in #40

closed 2024-10-17 by GDYendell