DiamondLightSource/fastcs #42: Add a helper for consistently creating PV names

GDYendell, 2024-06-18:
Currently PV names are formatted in multiple places and risk being inconsistent. There should be a single function that creates these PV names.

This may require refactoring the Attribute / Controller / Mapping structure to allow this function to have a simple interface. Currently the places that format PV names combine the path from the controller with the name of the attribute from the mapping.

Additionally the GUI needs to manually add the prefix, whereas the IOC does `builder.SetDeviceName`, so the prefix should probably be an attribute of `EpicsGUI` and applied while making the pvi signals to make this consistent.

GDYendell, 2024-09-02:
We don't call `builder.SetDeviceName` as of #54, so this should be slightly simpler.

evalott100, 2025-03-21:
To add some context:

### PVA
... currently converts from snake to pascal:

https://github.com/DiamondLightSource/FastCS/blob/1cce8ea7cbe0e7b31c9f294891f80bd8bbc60079/src/fastcs/transport/epics/pva/ioc.py#L25-L35

### CA and GUI
...currently convert `"_"` to `""`:

https://github.com/DiamondLightSource/FastCS/blob/1cce8ea7cbe0e7b31c9f294891f80bd8bbc60079/src/fastcs/transport/epics/ca/ioc.py#L117-L125

https://github.com/DiamondLightSource/FastCS/blob/1cce8ea7cbe0e7b31c9f294891f80bd8bbc60079/src/fastcs/transport/epics/gui.py#L42-L44

### We need

A method which all of these call which converts to PascalCase. We should error if there are characters other than alphanumerics or "_" in the input.

PR #144 (2025-05-23): Pv creation helper
Added a single helper function to convert PV names into Pascal case and replace both "-" and "_" characters.

I'm opening the PR to discuss if more actions might be necessary for the helper as it's currently not checking for any other characters (like "/", "%". ...) and it required a small change in the tests that I'm not entirely sure is correct.

Closes #42