DiamondLightSource/fastcs #53: Print a warning when PV names are too long instead of failing to start IOC

GDYendell, 2024-08-19:
When a PV name that is too long is passed to pythonSoftIOC it will raise and an exception. We should check the length before passing it in, printing a warning and ignoring the PV when it is too long.

## Acceptance Criteria
- IOC still starts successfully and excludes PVs that would have been too long

jsouter, 2024-09-06:
Created a branch for this https://github.com/DiamondLightSource/FastCS/tree/name_warning but didn't pull in the newer changes, happy to rebase on top of the pvi changes.

jsouter, 2024-09-06:
Draft PR here https://github.com/DiamondLightSource/FastCS/pull/56. Functional, but I will try to see if I can implement in a nicer way.

GDYendell, 2024-09-19:
Fixed by #56