DiamondLightSource/fastcs #62: Allow `Scan` to take no parameters

evalott100, 2024-10-15:
Currently `Scan` is only validating correctly if initialised through the `scan` decorator:

https://github.com/DiamondLightSource/FastCS/blob/bdab4fa51aaf8ac4459af8215d96e56651f4a30a/src/fastcs/cs_methods.py#L60-L61

This is because `inspect.Signature.parameters` will ignore `self`, but not in the wrapped case:

``` python
import inspect

class SomeClass:
    def __init__(self, fn, b):
        print(f"someclass fn {fn.__name__}", inspect.signature(fn).parameters)

def dec(b: str):
    def wrapper(fn):
        fn.wrapped_foo = SomeClass(fn, b)
    return wrapper

class X:
    def __init__(self):
        print("internal foo", inspect.signature(self.foo).parameters)
        print("internal bar", inspect.signature(self.bar).parameters)
        SomeClass(self.foo, "c")

    async def foo(self): ...
    async def bar(self, a: str): ...

    @dec("b")
    async def wrapped_foo(self): ...

x = X()
```

gives us
```
someclass fn wrapped_foo OrderedDict([('self', <Parameter "self">)])
internal foo OrderedDict()
internal bar OrderedDict([('a', <Parameter "a: str">)])
someclass fn foo OrderedDict()
```

I would like to be able to pass a poll period into my controller at init time

``` python
class PandaController(Controller):
    def __init__(self, hostname: str, poll_period: float) -> None:
        ...
        self.fastcs_method = Scan(self.update, poll_period)
```

but this leads to no `parameters` in the check above failing because the expected `self` arg isn't present.

# Suggestion

Change

https://github.com/DiamondLightSource/FastCS/blob/bdab4fa51aaf8ac4459af8215d96e56651f4a30a/src/fastcs/cs_methods.py#L60-L61

to

``` python
        if self.parameters > 1 or (self.parameters and tuple(self.parameters.keys()) != ("self",)
            raise FastCSException("Scan method cannot have arguments")
```

GDYendell, 2024-10-15:
It is intended that methods of `Controller`s have `fastcs_method` added so that they get processed as `Method`s by the backend, not to add it to the `Controller` itself. What are you hoping to happen by adding `self.fastcs_method` to `Controller`?

evalott100, 2024-10-15:
It was my understanding that on `Backend.__init__`, `Mapping` is initialised and fields in the `Controller` wrapped in `Scan` would be picked up in the walk and have their scan set up in `Backend._start_scan_tasks`?

My aim is really just to have that method be picked up as a `Scan` this, but without using the decorator since `--poll-period` being an argument means it has to be passed in on the child `Controller` init.

GDYendell, 2024-10-16:
Ah I see. No it does not find fields that are `Scan`, it finds methods that have an attribute `fastcs_method` that is a `Scan` ([here](https://github.com/DiamondLightSource/FastCS/blob/main/src/fastcs/mapping.py#L50)).

So, if you want to do this dynamically from a CLI argument you would need to make a method (or a Callable) in `__init__` with the ScanCallback signature and attach a Scan(dynamic_period) to it. See how that goes. If that doesn't work or is horrible, we might need to rethink the API.

evalott100, 2024-10-16:
My initial attempt was `self.fastcs_method = Scan(self.update, poll_period)` in the controller `__init__` before `super().__init__` and this did seem to work until the validator saw that `self.update` doesn't have `self` in parameters (which it currently expects because of the signature behavior above).

evalott100, 2024-10-16:
Initially I misread and did something like

``` python
self.fastcs_method = Scan(self.update, poll_period)
```

in `__init__` of my controller, however going by the wrapper I should instead be adding the attribute `fastcs_method` to the `update`, however you can't do that on an object method, only the class... so

``` python
# Globally
PandaController.update.fastcs_method = Scan(PandaController.update, 0.1)
```

Works and sets up the scan correctly - the signature parameters see `self` `len(parameters) == 1`.

GDYendell, 2024-10-16:
Can you not just do `self.update.fastcs_method = poll_period` in `__init__`?

evalott100, 2024-10-16:
> Can you not just do `self.update.fastcs_method = Scan(self.update, poll_period)` in `__init__`?

Unfortunately not, you can't add attributes to methods on objects, only methods on classes:

```
from pprint import pprint
class Foo:
    def bar(self):
        pass

setattr(Foo.bar, "some_string", "meh")
setattr(Foo().bar, "some_string", "meh") # Raises attribute error
```

GDYendell, 2024-10-16:
Ah, right. You can do
```
setattr(Foo().__class__.bar, "some_string", "meh")
```

but that is pretty horrible.

I have recently tried refactoring this so that the decorator replaces the method with a callable `Scan`, but I still couldn't convince pyright strict, so I shelved that for now. But maybe it would be an improvement even if it doesn't satisfy pyright. Then I think you could do

```
self.update = Scan(self.update, period)
```

GDYendell, 2024-10-16:
I have reworked my WIP [here](https://github.com/DiamondLightSource/FastCS/tree/dynamic-methods). See the [test](https://github.com/DiamondLightSource/FastCS/blob/dynamic-methods/tests/test_controller.py#L38) for an example of creating a `BoundCommand` from an instance method. `BoundScan` should work in the same way.

closed 2025-03-10 by GDYendell