DiamondLightSource/fastcs #63: Allow further epics options in `datatypes`

evalott100, 2024-10-15:
Currently you can pass general information to the datatypes, e.g `Bool(znam="1", onam="1")` or `Float(prec=0)`. We need to be able to add epics specific information too, for example `EGU`, `DESC`.

I suggest that we create a new optional kwarg on either the datatypes or attributes:`configuration=`. This will be an child of a new abstract class to describe further information about the datatype. Then in the epics backend we can check for this (an `EpicsIntConfiguration` and set our sub fields on the record accordingly).

@GDYendell

evalott100, 2024-10-16:
I suppose egu will have to live in `Float` and `Int` since it's an `aIn`/`aOut`/`longIn`/`longOut` specific field. I'll make a new branch and use my best judgement as to what should live where.

GDYendell, 2024-10-16:
I think we shouldn't couple it to what EPICS does - backend would still be able to only pass through egu for the record types that support it. But yes, in this case egu on string and bool would never make sense, so I think you are right it needs to go on the type.

GDYendell, 2025-08-06:
This is done and the EPICS CA transport is implementing description. Just units needs to be used for EGU [here](https://github.com/DiamondLightSource/FastCS/blob/main/src/fastcs/transport/epics/ca/util.py#L53).