DiamondLightSource/fastcs #64: Allow different pv naming convensions as input options

evalott100, 2024-10-17:
In https://github.com/DiamondLightSource/FastCS/issues/42

All attribute names were converted to pascal case. We should add a new options enum to pass into the ioc specifying what conversion to do, and specifying the epics seperator.

GDYendell, 2024-10-22:
I am happy with what you have on your branch so far, but would like either
- sub controller node and leaf naming conventions to specified separately (e.g. caps for nodes and pascal for leaf)
- don't convert node names at all and just use the string passed to register_sub_controller - we only need to convert the attribute names really as they have to defined as python attribute names

evvaaaa, 2025-09-17:
> I am going to close this for now and say the PV formatting should not be configurable. If we have a case where this is absolutely necessary we can consider it.

Yup, in hindsight fastcs should always be forcing one particular naming convention IMO. Hope all is going well with development :)

Tom-Willemsen, 2026-01-13:
Thanks - if you're open to it in principle, I'll try and create a draft PR for something in this sort of shape at some point.