DiamondLightSource/fastcs #65: Allow sub controllers to have a different label to their path name

GDYendell, 2024-10-18:


GDYendell, 2024-10-22:
This might not be possible because qsrv / pvaccess does not allow fields starting with a number, so the PVI info creation fails:
```
ODIN:FP:PVI: Error Group not created: invalid field name "0"
ODIN:FR:PVI: Error Group not created: invalid field name "0"
```

evalott100, 2024-10-24:
https://github.com/DiamondLightSource/FastCS/commit/111287a26a39ebcb5ec2da19dd6f32901ef4afa9 seems to work, I'll make it better when I've decided what I'm doing.

```
FASTCS_PANDA:TTLIN1:PVI_PV
FASTCS_PANDA:TTLIN1:Term
FASTCS_PANDA:TTLIN1:Term_RBV
FASTCS_PANDA:TTLIN2:PVI_PV
FASTCS_PANDA:TTLIN2:Term
FASTCS_PANDA:TTLIN2:Term_RBV
FASTCS_PANDA:TTLIN3:PVI_PV
FASTCS_PANDA:TTLIN3:Term
FASTCS_PANDA:TTLIN3:Term_RBV
FASTCS_PANDA:TTLIN4:PVI_PV
FASTCS_PANDA:TTLIN4:Term
FASTCS_PANDA:TTLIN4:Term_RBV
FASTCS_PANDA:TTLIN5:PVI_PV
FASTCS_PANDA:TTLIN5:Term
FASTCS_PANDA:TTLIN5:Term_RBV
FASTCS_PANDA:TTLIN6:PVI_PV
FASTCS_PANDA:TTLIN6:Term
FASTCS_PANDA:TTLIN6:Term_RBV
```