DiamondLightSource/fastcs #74: @command does not work in python 3.12

dperl-dls, 2024-11-13:
The resulting function fails an `isinstance` for `WrappedMethod` so `_get_single_mapping` does not add it to the `Mapping`

## Steps To Reproduce
Steps to reproduce the behaviour:
1. Create a `Controller` with a `@command` method in python 3.12
2. Observe that there is no PV created for it

## Acceptance Criteria
- The `@command` decorator can be used in python 3.12

GDYendell, 2024-11-14:
Thanks @dperl-dls ! Currently we are pinned <3.12 because softioc was not compatible, but I think that is resolved now, so we should unpin and fix what breaks.

dperl-dls, 2024-11-14:
Yeah, most stuff does just work, I was running on 3.12 generally successfully and got surprised when my commands were not resulting in generated PVs in the IOC. I tried running the tests yesterday when writing this issue and only a couple failed

closed 2025-03-10 by GDYendell