DiamondLightSource/fastcs #85: Add `WaveForm` and `Enum` dataypes

evalott100, 2024-11-20:
## `WaveForm`

Will be either a `Sequence[T]` where `T` is `float`, `int`, or `str`, or a numpy array with a `dtype`. This will not be an arg to the dataclass (not optional).

There will be a `length` on the datatype set to some sane default, since this is required by the EPICS and [tango backends](https://tango-controls.readthedocs.io/en/latest/tutorials-and-howtos/how-tos/how-to-deal-string-tango-attribute-cpp.html#spectrum-image-attribute-static-allocation). 

## `Enum`

Will take a python `enum.Enum` type. When putting to the attribute you put the enum member not a string/int value itself. We will check on initialisation that all the values are either string or int.

We should keep allowed values for checking on `Attribute.set`, but I don't think that the current system of making and `mbb` record if it's a string with allowed values is as nice as having a dedicated `Enum` datatype that can be altered to whatever control system's enum definition.

## Changes which will be required across datatypes.

`DataType` will now be given two additional attributes:

* `cast(T)`: For casting to a datatype which can be sent over the control system. It will also do some validation (i.e checking that an `Int` value doesn't exceed the maximum defined in the `DataType`).

``` python
WaveForm(Sequence[str]).cast(["a"]) == np.array(["a"], dtype="S40")
Enum(SomeEnum).cast(SomeEnum.member) -> "TheStringValueOfTheMember"
# Fails so that we don't have an internal value of the `Attribute` out of sync with the softioc.
Int(max=5).cast(6)
```

* `@property\intitial_value()`: We can't rely on `Attr._datatype.dtype()` for an initial value anymore since the structured numpy array's has to be initialised with shape (length in our case) and `dtype=`. For enums this is true too. Therefore, we will use `DataType.initial_value` for the default initial value on attributes. This can be defaulted to `self.dtype()` in the `Attribute` base class.

GDYendell, 2024-11-20:
I think `Enum.cast` should be `Enum.validate` using the mechanism suggested [here](https://github.com/DiamondLightSource/FastCS/pull/82#discussion_r1850143135).

GDYendell, 2024-11-20:
> We should keep allowed values for checking on `Attribute.set`

Can we just do `self.dtype.validate(value)` instead of keeping `allowed_values` on `Attribute`? Actually, I don't think it even validates against `allowed_values` currently.

evalott100, 2024-11-20:
> I think `cast` should be `validate` using the mechanism suggested [here](https://github.com/DiamondLightSource/FastCS/pull/82#discussion_r1850143135).

Okay, I'll move it to that PR.

closed 2025-01-31 by GDYendell