DiamondLightSource/fastcs #89: Proposal for changing how we search for attributes

evalott100, 2024-11-20:
Currently, FastCS [walks all attributes](https://github.com/DiamondLightSource/FastCS/blob/4c9d82a7dd921c8b5d202514ec39f28a2451b92c/src/fastcs/mapping.py#L40-L56) in the `dir` of controller to find attributes, which can be a problem for two reasons.

1. If the name of an attribute is dynamic then during `initialise()` the controller will have to 

``` python 
dynamic_name, dynamic_type = ...
setattr(dynamic_name, however_it_generates_attributes(dynamic_type))
```

2. There are some places in pandablocks where we want a controller which also has a subcontroller, but also a top level attribute provided by the subcontroller:

``` class Controller2(SubController):
        def __init__(self):
            self.sub_attribute = AttR(Int())
            self.attribute_for_base_controller = AttR(Float())

    class Controller1(Controller):
        async def initialise(self):
            block = Controller2()
            self.block = block.attribute_for_base_controller
            self.register_sub_controller("block", block)
```

We want `Controller1` in the above to have PVs

```
PREFIX:BLOCK - a float
PREFIX:BLOCK:PVI
PREFIX:BLOCK:SUB_ATTRIBUTE - an int
```
instead we'd have to wrap `attribute_for_base_controller` in some structure so that it isn't seen as an `Attribute` in the mapping.

### Suggestion
There are three ways to define attributes on a controller:

``` python
class SomeController(Controller):
    some_attr_annotation: AttrR[int] # 1
    class_attr = AttrR(Float()) # 2
    
    def __init__(self):
        self.some_attr_annotation = AttrR(Int())

    def get_attributes(self) -> dict[str, Attribute]:
        return {"some_attr": AttrR(String())} # 3
```

Walk mappings will now:

### 1/2
Look for attributes in type(controller).__annotations__. This will include annotations `#1` and `#2`. In both cases we get the attribute from `getattr(controller, annotation_name)`.

In the case of `#1` we raise an error if the annotated attribute doesn't show up in `controller` by the time we walk mappings (post `initialise`).

In the case of `#2` we're already handling mutability so that `class_attr` won't be changed when `controller.class_attr` changes it's state.

Principally, we're checking for the `Attributes` in `type(controller)`, and not `controller`. This lets a controller make `Attributes` which it can put in other controllers (defined in `__init__`).

### 3
We allow the user to provide whatever additional attributes via a method, defined to return `{}` in the `BaseController`.

evalott100, 2024-11-20:
@GDYendell @jsouter

GDYendell, 2024-11-21:
I think this seems reasonable. I am still a little unclear what the use cases (current and future) for passing attributes up to the parent controller might be.

For panda, is there only ever a single Attribute in the parent for a given child controller, or could there be multiple? What is the value of the attribute, what does it represent?

evalott100, 2024-11-21:
In panda we have quite a lot of places in fields where we do this:

```
        structure out
            string r PANDA1:COUNTER1:OUT
        structure out_capture
            string rw PANDA1:COUNTER1:OUT:CAPTURE
        structure out_dataset
            string rw PANDA1:COUNTER1:OUT:DATASET
        structure out_mode
            string rw PANDA1:COUNTER1:OUT_MODE
        structure out_offset
            string rw PANDA1:COUNTER1:OUT:OFFSET
        structure out_scale
            string rw PANDA1:COUNTER1:OUT:SCALE
        structure out_units
            string rw PANDA1:COUNTER1:OUT:UNITS

...
        structure trig
            string rw PANDA1:COUNTER1:TRIG
        structure trig_delay
            string rw PANDA1:COUNTER1:TRIG:DELAY
        structure trig_edge
...
        structure dir
            string rw PANDA1:COUNTER1:DIR
        structure dir_delay
            string rw PANDA1:COUNTER1:DIR:DELAY

```

> For panda, is there only ever a single Attribute in the parent for a given child controller, or could there be multiple?

The subcontroller will only ever pass one attribute element back to the controller to act as the `root_element`.

If we wanted to more formally support this we could also add a property

```
SubController::root_attribute() -> Attribute | None
```

which would be used in the parent controller `_get_single_mapping` to create a `PREFIX:BLOCK` attribute for a controller `PREFIX:BLOCK:PVI`.

GDYendell, 2024-11-21:
If it is just one then a property would be a clearer API. Saying that, if there were more than one it could be a `root_attributes` property that returns a list.

I do still think we should make some of these changes to the way we search for attributes, though.

PR #91 (2024-12-04): Change how we search for attributes
Closes #89
closed 2024-12-04 by GDYendell