DiamondLightSource/fastcs #122: Allow for explicit pairing of SubControllers/Attributes

evalott100, 2025-02-13:
In https://github.com/DiamondLightSource/FastCS/pull/119, we pair attributes/controllers together based on the name of the attribute. If the name of two attributes/subcontrollers `foo_1`/`foo_2` or `Foo1`/`Foo2` is the same then they're grouped together in the same pvi group under `foo`.

# Problem

This is assuming too much on the FastCS side. We should instead allow controllers to have a `AttributeVector`/`ControllerVector` called `foo` which can be indexed by `int` (not necessarily contiguous). On the transport, these elements are grouped together under `foo`.

# Required Changes

1. Add a new `AttributeVector` type.
2. Add a new `ControllerVector` which can be passed into `register_subcontroller`.

evalott100, 2025-02-19:
> Could you outline how PviTree would change.

We wouldn't be assuming `Device:Foo1` and `Device:Foo83` should exist as

```
foo (structure):
    d (structure):
        v1 Device:Foo1:PVI
        v83 Device:Foo83:PVI
```

Simply because they both have the same end string up to the number. We would have a `SubControllerVector` in the Device `Controller` called `Foo` with entry `1` and `83`, and the above structure would be obtained, or if we just register them as controllers like normal we'd have

```
foo1 (structure):
    d Device:Foo1:PVI
foo83 (structure):
    d Device:Foo83:PVI
```

GDYendell, 2025-02-20:
@evalott100 re: our very short discussion as I ran out the door. After our session today, I understand your last comment. Would it make sense to rename this issue something like "Add SubControllerVector to allow grouping multiple instances of a SubController".

Ophyd has a `DeviceVector` - does it also have a `SignalVector`? Does PandA have a use case for `AttributeVector` or just `SubControllerVector`? I think I might have a use case for it to simplify something in fastcs-odin. But, if you don't need it right now maybe it could be done separately later?

evalott100, 2025-02-21:
Will close this and write a new issue with these changes.

# 1

> "Add SubControllerVector to allow grouping multiple instances of a SubController".

Sure, I think we should also include the ophyd-async style Controller parsing we were discussing:

* All `Attribute`s which are on the `controller` object as python attributes get added to `controller._attributes: dict[str, Attribute]`
* All `Controller`s which are on the `controller` object as python attributes get added to `controller._sub_controllers: dict[str, Controller | ControllerVector]`

# 2

I don't really see any need for sub controllers to be their own class `SubController`? I think we should remove it. A *subcontroller* is just a controller living in another controller.

# 3

> If you don't need it right now maybe it could be done separately later?

Yep I agree, we should only cover `class ControllerVector(dict[int, Controller | ControllerVector])`, we could make this a subclass of `Controller`/`BaseController` too if it ends up being more convenient, especially if we do decide to get rid of `SubController`.

If a transport has no interpretation of `ControllerVector` then it just uses each of the contained `Controller` values as normal from `get_flattened_sub_controllers:

``` python
class Controller(BaseController):
    def get_sub_controllers(self) -> dict[str, Controller | ControllerVector]:
        return self._controllers

    def get_flattened_sub_controllers(self) -> dict[str, Controller]:
        # uses sub_controller.flatten() if sub_controller is a ControllerVector
```

GDYendell, 2025-02-21:
I _think_ the difference between Controller and SubController is purely philosophical at this point, but there may be some practical details that we will need to abstract out.

evalott100, 2025-03-07:
# UPDATE

We shouldn't group `Attribute`s at all. This is incorrect in the current p4p backend.