DiamondLightSource/fastcs #125: Search Controller for GenericAlias of Attributes

GDYendell, 2025-03-07:
It should be possible to define Attributes like:

```python
class MyController(Controller):
    attr: AttrRW[int]
```

Currently these are ignored because they fail `isinstance(attr, Attribute)`, because they are `GenericAlias`

`Controller._bind_attrs` should have an additional case to check if attributes are `GenericAlias` with an `Attribute` as `__origin__`. If so the `Attribute` can be instantiated by reading the datatype from `__args__`.

## Acceptance Criteria
- Attributes can be instantiated as in the example above
- Add a check to `test_backend.py::test_controller_api` that these attributes are included in `attributes`.

jsouter, 2025-08-11:
If we hint an attribute that we expect to be dynamically created during `<Base>Controller.initialise()` then the first attribute created in `_bind_attrs` will get overwritten, is that fine? Maybe we use typing.Annotated to provide an arg for if we want the hinted attribute to be created in `_bind_attrs`, also can pass in more args than just the datatype if we need a custom handler.

jsouter, 2025-08-11:
I think that makes more sense - we can always add in support for Annotated later (though I don't think there's much of a need for that).

GDYendell, 2025-08-13:
Closing to avoid complications with #124

PR #187 (2025-08-14): assert hinted attrs are bound and have right dtype after initialisation
~Closes #125~
Closes #124
closed 2025-08-13 by GDYendell