DiamondLightSource/fastcs #129: Validating datatypes

jsouter, 2025-03-10:
Working on porting fastcs-eiger to FastCS@main, have come up with a few points so far:

- setting an attribute with a Float() datatype fails if we call `await attribute.set(value)` with an int. Should this cast to float?
- Enum attributes must be set with instances of the Enum class, this means when parsing the value of a string with allowed_values from the response to a GET from the eiger we have to keep track of the Enum class and cast the value, should we allow the raw value to be passed to set if it one of the values in the Enum class?

evalott100, 2025-03-11:
> setting an attribute with a Float() datatype fails if we call await attribute.set(value) with an int. Should this cast to float? (possibly this points to an issue with the tickit sim, but I imagine this issue will arise elsewhere).

This was a concious choice when I made the changes to `DataType`. If it's possible that the `Float` attribute is receiving either `int` or `float` (as python primatives) then it's the job of downstream to convert the `int` to the `float`.

I would be open to changing so that `Float.validate` performs the conversion `int -> float`, but I think that it should still be typed as

``` python
Float.validate(value: float) -> float
```

rather than

``` python
Float.validate(value: float | int) -> float
```

so doing this still gives a linting error. I still think I prefer for this to be handled downstream though. `Float` will do coercion for precision, but that's only because it's a field that has meaning to the transports.

> Misunderstood how enums work in pva, will see if i can fix on the fastcs-eiger end but this may still need addressing

You should be able to just `datatype.enum_cls["ENUM_MEMBER_NAME"]` or `datatype.enum_cls("ENUM_MEMBER_VALUE")`.

I don't think we should provide any way of passing in the value or name to an `Enum` `Attribute`. I think this would sort of defeat the purpose of there being an enum in the first place, and also lead to confusing situations where downstream thinks they can pass `0` to mean the first element of a string enum, when it's just an implementation detail that the pva transport uses the index of the enum rather than the value.

jsouter, 2025-03-11:
🫡 happy to handle downstream