DiamondLightSource/fastcs #130: `Sender`/`Updater`/`Handler` now incorrectly take the root controller

evalott100, 2025-03-10:
`Sender`, `Updater`, and `Handler` is now broken for subcontrollers. The `controller` passed as an arg to `update` and `put` is always the main `Controller` passed into FastCS, rather than the controller the attribute is registered under.

evalott100, 2025-03-10:
This is because the `controller` passed in here:

https://github.com/DiamondLightSource/FastCS/blob/759fa13f223ec014719b98b45d075d8edeb5673f/src/fastcs/backend.py#L76-L78

and here:

https://github.com/DiamondLightSource/FastCS/blob/759fa13f223ec014719b98b45d075d8edeb5673f/src/fastcs/backend.py#L117-L120

are always the root controller.

jsouter, 2025-03-10:
Can do something like

```
def _get_subcontroller_from_path(path: list[str], controller):
    if not path:
        return controller
    subcontroller = controller.get_sub_controllers()[path[0]]
    return _get_subcontroller_from_path(path[1:], subcontroller)
````
and then use
`controller = _get_subcontroller_from_path(controller_api.path, root_controller)` when looping through `controller_api.walk_api()`

evalott100, 2025-03-10:
> but bit of a dirty fix

It does feel pretty nasty to pull out everything FastCS needs into the `ControllerAPI` and then look up controllers for that one piece of information.

I say storing the callback in `ControllerAPI` while it's being build is nicer.

GDYendell, 2025-03-10:
Yeah, I think `_build_controller_api` should become an instance method of `Backend` so that it can create the tasks and the `ControllerAPI` at the same time. I am just playing around with this now to see if it comes together nicely.

PR #135 (2025-05-19): Add Handler.initialise(controller)
Resolves #59 
Added default initialise methods to Sender and Updater base classes.
Added initialise method to Attribute, checks for handler and calls the method.
Moved initialise into BaseController class, loops over attributes calling initialise with a reference to self.
Added unit test for Attribute class.

Fixes #59 
Fixes #130