DiamondLightSource/fastcs #149: Enum choices

jsouter, 2025-04-02:
Currently enum choice values are restricted to valid python names, as the choices are populated from the enum member names.
https://github.com/DiamondLightSource/FastCS/blob/2c546da833d390bf32806258acdc282c55c8b669/src/fastcs/transport/epics/ca/util.py#L71-L79

This restricts us from being able to have spaces in strings and other characters which would otherwise normally be allowed in ZRST, ONST, TWST,... in EPICS. I can understand why this is the case, but it would be nice if this could be supported. 

As a corollary, 
https://github.com/DiamondLightSource/FastCS/blob/2c546da833d390bf32806258acdc282c55c8b669/src/fastcs/transport/epics/ca/util.py#L85-L105

Enums with more than 16 values don't work correctly in Channel Access, I believe we just need to add extra logic to handle casting to/from epics types if `len(attribute.datatype.members) > MBB_MAX_CHOICES`. 
In Channel Access, `Enum`s with too many choices decay to `longin`/`longout`.

GDYendell, 2025-08-06:
Are we happy that this can be worked around by constructing an Enum with a dict @jsouter ?

jsouter, 2025-08-06:
> Are we happy that this can be worked around by constructing an Enum with a dict [@jsouter](https://github.com/jsouter) ?

Yes, I think so

closed 2025-08-07 by GDYendell