DiamondLightSource/fastcs #152: Phoebus screens issues with PVA

jsouter, 2025-04-03:
I may have something misconfigured, but when comparing fastcs-eiger with EpicsCAOptions and EpicsPVAOptions there are some issues with PVA.

Firstly, by default all PVs appear disconnected.
As a hack, if I prefix `pva://` to the return value of
https://github.com/DiamondLightSource/FastCS/blob/2c546da833d390bf32806258acdc282c55c8b669/src/fastcs/transport/epics/gui.py#L42-L44

Controls/monitors for R and W type `Attribute`s work, but since we don't create `_RBV` PVs in PVA, we are rendering extra monitors that shouldn't exist.

[image]

GDYendell, 2025-04-04:
Using pva PVs should probably be supported in pvi. I don't know if this is possible by setting some root property in the bob file that the default is pva:// instead of ca://, or if it would require prefixing every PV.

> but since we don't create _RBV PVs in PVA

Why is this?

jsouter, 2025-04-04:
> Why is this?

I'm not sure, but the disparity currently exists between the CA and PVA transports. I wasn't sure if it was an oversight or an intentional choice.

GDYendell, 2025-04-04:
I have chatted with Tom and Gareth. We think PVA should also have RBV fields. Could you make an issue to do this and in the acceptance criteria state that we need to write an ADR to explain why.